pauldemarco/flutter_blue #1003: loss data

oooldCaption, 2021-12-16:
I have a subscription with two types of data, one with a frequency of 100hz and one with a frequency of 50hz. But in 20 seconds, I only received 1700~1800 sets of data, which should be around 3000 data.

I use nrf connect to view the log, and the data in 20 seconds is 2800~3000;
```
_bleDeviceDataUpload() async {
    /// valueCharacteristic  ->  notifications and listen to changes
    valueCharacteristic.setNotifyValue(true);
    valueCharacteristic.value.listen((value) {
      if (value == null || value.length == 0) {
        return;
      } else {
        print("------$value");
      }
    });
  }

```
notifications data like this 
```
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 4, 11, 0, 0, 118, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 8, 11, 0, 0, 114, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 9, 11, 0, 0, 122, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 254, 10, 0, 0, 121, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 2, 11, 0, 0, 118, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 5, 11, 0, 0, 117, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 5, 11, 0, 0, 109, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 3, 3, 4, 11, 0, 0, 117, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
********[3, 0, 0, 2, 8, 0, 0, 4, 2, 134, 255, 114, 255, 92, 255, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
......
.....
```

flutter doctor
```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.5.3, on macOS 12.0.1 21A559 darwin-arm, locale zh-Hans-CN)
[✓] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
[✓] Xcode - develop for iOS and macOS
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[☠] IntelliJ IDEA Ultimate Edition (the doctor check crashed)
    ✗ Due to an error, the doctor check did not complete. If the error message below is not helpful, please
      let us know about this issue at https://github.com/flutter/flutter/issues.
    ✗ FormatException: Unexpected extension byte (at offset 5)
[✓] IntelliJ IDEA Ultimate Edition (version 2021.3)
[✓] VS Code (version 1.64.0-insider)
[✓] Connected device (2 available)
    ! Error: iPhone is not connected. Xcode will continue when iPhone is connected. (code -13)

! Doctor found issues in 1 category.
```
flutter_blue : 0.8.0

oooldCaption, 2021-12-16:
I read the source code and found this problem. Every time the data is updated, a new object is created, which may affect the efficiency. I modified the source code:
like this , it's work. no loss data:
```
src/main/java/pauldemarco/flutter_blue/FlutterBluePlugin.java 

        Protos.OnCharacteristicChanged.Builder vp = Protos.OnCharacteristicChanged.newBuilder();
        @Override
        public void onCharacteristicChanged(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic) {
            log(LogLevel.DEBUG, "[onCharacteristicChanged] uuid: " + characteristic.getUuid().toString());
            Protos.OnCharacteristicChanged.Builder vp = Protos.OnCharacteristicChanged.newBuilder();
            vp.setRemoteId(gatt.getDevice().getAddress());
            vp.setCharacteristic(ProtoMaker.from(gatt.getDevice(), characteristic, gatt));
            invokeMethodUIThread("OnCharacteristicChanged", vp.build().toByteArray());
            // Protos.OnCharacteristicChanged.Builder p = Protos.OnCharacteristicChanged.newBuilder();
            // p.setRemoteId(gatt.getDevice().getAddress());
            // p.setCharacteristic(ProtoMaker.from(gatt.getDevice(), characteristic, gatt));
            // invokeMethodUIThread("OnCharacteristicChanged", p.build().toByteArray());
        }
```
[image]

closed 2021-12-16 by oooldCaption

MrCsabaToth, 2021-12-25:
Ref https://github.com/PhilipsHue/flutter_reactive_ble/issues/481

MrCsabaToth, 2021-12-25:
So you hoisted the `Protos.OnCharacteristicChanged.newBuilder()` creation from inside the method to class level? I wonder what long term consequences does this have. This pattern (of creating the Proto objects per method call) is used in all the methods so one has to wonder there must be a reason for that.

oooldCaption, 2021-12-25:
> So you hoisted the `Protos.OnCharacteristicChanged.newBuilder()` creation from inside the method to class level? I wonder what long term consequences does this have - by long term I mean application lifecycle wide. This pattern (of creating the Proto objects per method call) is used in all the methods so one has to wonder there must be a reason for that.
> 
> 所以您将 protos.oncharacterchanged.newbuilder ()创建从方法内部升级到类级别？我想知道这样做的长期后果是什么——我指的是整个应用程序生命周期的长期后果。这种模式(每个方法调用创建 Proto 对象)被用于所有的方法，所以我们不得不怀疑这一定有原因。

OnCharacteristicChanged aways called every time a notification is received, so,  150 objects will be created in 1 second. It should be a very time-consuming operation.  maybe it should be a singleton object

MrCsabaToth, 2021-12-25:
I only receive data 2-6 times per second, but on old devices the connection hangs after 5-10 minutes and I have a feeling that this could be the cause?

oooldCaption, 2021-12-25:
> I only receive data 2-6 times per second, but on old devices the connection hangs after 5-10 minutes and I have a feeling that this could be the cause?
> 
> 我每秒只能收到2-6次数据，但在旧设备上，连接会在5-10分钟后挂起，我觉得这可能就是原因？

😂😂😂  I haven't encountered a hangs;  Do you think it is because too many objects occupy too much memory, causing the application to be hangs?

MrCsabaToth, 2021-12-25:
I haven't identified how much the UI update takes part in that bug, and how much just the BLE data processing, but it happened. During the holidays hopefully I'll have time to debug it more. Theoretically the JVM (or rather Dalvik) garbage collection should clean up those `Protos.OnCharacteristicChanged` objects, so I would not think about memory problem, but I'd have to profile it to confirm. Note that my testing device is Android 6.0, quite old + slow. But some of my users have such old devices.

MrCsabaToth, 2021-12-25:
Another thing: about the class level static `Protos.OnCharacteristicChanged.Builder vp = Protos.OnCharacteristicChanged.newBuilder();`. What if I connect to two different BLE devices at the same time, both of them are sending 2-6 data packets per second. Can there be any race condition arise from those two asynchronous independent event sources to use that single shared static object? Is that static object thread-safe?

oooldCaption, 2021-12-27:
> 

Just discard the data. The app does not tell the Bluetooth device how much data the app receives

> Another thing: about the class level static `Protos.OnCharacteristicChanged.Builder vp = Protos.OnCharacteristicChanged.newBuilder();`. What if I connect to two different BLE devices at the same [...]

👍👍👍
good question. You are very keen.