pauldemarco/flutter_blue #1006: cannot install, conflict with other plugin.

eggysudianto, 2021-12-22:
cannot install, conflict with other plugin.  can you please update `rxdart` to version `^0.27.3`?

```
Because flutter_blue >=0.8.0 depends on rxdart ^0.26.0 and flutter_form_bloc >=0.28.0-alpha depends on rxdart ^0.27.3, flutter_blue >=0.8.0 is incompatible with flutter_form_bloc >=0.28.0-alpha.
So, because otlink depends on both flutter_form_bloc ^0.29.3 and flutter_blue ^0.8.0, version solving failed.
pub get failed (1; So, because otlink depends on both flutter_form_bloc ^0.29.3 and flutter_blue ^0.8.0, version solving failed.)
```

peekpt, 2021-12-22:
https://github.com/pauldemarco/flutter_blue/issues/979

boskokg, 2022-01-31:
Please go to flutter_blue_plus and report issues there if still exist.