pauldemarco/flutter_blue #1007: always disconnect after connected to device

yusmi25, 2021-12-22:
Hey I'm using flutter_blue 0.8.0, but always disconnect after connected to device
this is my code 
`
try {
    await speedDevice.connect();
  } catch (e) {
    print("error Connect Speed : ${e.toString()}");
  }
  speedDevice.state.listen((event) async {
    if (event == BluetoothDeviceState.connected) {
      try {
        //discover service
        } catch (e) {
        print("error connect Discover service : ${e.toString()}");
      }
    }
  });
`
this is log console
I/BluetoothAdapter(21465): STATE_ON
D/BluetoothGatt(21465): connect() - device: E5:BD:43:E0:A1:3D, auto: true
I/BluetoothAdapter(21465): isSecureModeEnabled
D/BluetoothGatt(21465): registerApp()
D/BluetoothGatt(21465): registerApp() - UUID=1625bc9d-5153-456a-aa53-1dc779504005
D/BluetoothGatt(21465): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt(21465): onClientConnectionState() - status=0 clientIf=9 device=E5:BD:43:E0:A1:3D
D/FlutterBluePlugin(21465): [onConnectionStateChange] status: 0 newState: 2
D/BluetoothGatt(21465): cancelOpen() - device: E5:BD:43:E0:A1:3D
D/BluetoothGatt(21465): onClientConnectionState() - status=0 clientIf=9 device=E5:BD:43:E0:A1:3D
D/FlutterBluePlugin(21465): [onConnectionStateChange] status: 0 newState: 0
D/BluetoothGatt(21465): close()
D/BluetoothGatt(21465): unregisterApp() - mClientIf=9
I/flutter (21465): ada error connect Discover service : Exception: Cannot discoverServices while device is not connected. State == BluetoothDeviceState.disconnected
thanks

thusithadeepal9356, 2022-05-03:
I have the same issue,do you fixed it?

dan-developer, 2023-01-01:
Same issue here +1

elmar001, 2023-07-10:
Same issue. Any solution?