pauldemarco/flutter_blue #1009: Wrong data after reconnecting to the same device.

GorIvanov, 2022-01-06:
There is a device that sends data...

So, after first connection it works well and in **event listener** each **event** data is new as should be.
Like:
[49, 5, 0, 0, 0, 13, 60, 169, 5, 133, 212, 0, 0, 0, 5, 159, 32, 0, 74, 92]
[50, 5, 252, 194, 139, 7, 44, 67, 0, 0, 0, 13, 60, 176, 5, 135, 149, 0, 0, 0]
[51, 5, 5, 162, 106, 0, 75, 240, 252, 195, 235, 7, 44, 175, 0, 0, 0, 13, 59, 208]

But! After **calling disconnect** and **connecting again**, it sends the same data several times.

[98, 1, 164, 148, 56, 161, 93, 50, 0, 0, 0, 163, 64, 149, 254, 92, 141, 0, 0, 0]
[98, 1, 164, 148, 56, 161, 93, 50, 0, 0, 0, 163, 64, 149, 254, 92, 141, 0, 0, 0]

[99, 1, 46, 54, 117, 73, 1, 35, 53, 181, 244, 9, 138, 92, 0, 0, 0, 29, 47, 15]
[99, 1, 46, 54, 117, 73, 1, 35, 53, 181, 244, 9, 138, 92, 0, 0, 0, 29, 47, 15]

`
//
_notificationService?.value.listen((event) {
      if (event.isNotEmpty) {
        print("EVENT $event");  // so, here I got the events with the same data after reconnecting.
      }
    });
//
`
At the same time, if you connect to some other device, the data will be in order until we disconnect and connect again.
Of course, if I close and reopen the application, I can reconnect to these devices again.

It was tested on Android and ios works same.
Maybe there are some buffers not cleared in the plugin for a specific device?

GorIvanov, 2022-01-06:
I have found only one solution to compare new even and the last event: 
[image]

oooldCaption, 2022-01-12:
> There is a device that sends data...
> 
> 有一个设备可以发送数据..。
> 
> So, after first connection it works well and in **event listener** each **event** data is new as should be. Like: [49, 5, 0, 0, 0, 13, 60, 169, 5, 133, 212, 0, 0, 0, 5, 159, 32, 0, 74, 92] [50, 5, 252, 194, 139, 7, 44, 67, 0, 0, 0, 13, 60, 176, 5, 135, 149, 0, 0, 0] [51, 5, 5, 162, 106, 0, 75, 240, 252, 195, 235, 7, 44, 175, 0, 0, 0, 13, 59, 208]
> 
> 因此，在第一次连接之后，它工作得很好，在事件侦听器中，每个事件数据都应该是新的。比如: [49,5,0,0,0,13,60,169,5,133,212,0,0,0,0,5,159,32,0,74,92][50,5,252,194,139,7,44,67,0,0,0,13,60,176,5,135,149,0,0,0,0][51,5,5,162,106,0,75,240,252,195,235,7,44,175,0,0,0,13,59,208]
> 
> But! After **calling disconnect** and **connecting again**, it sends the same data several times.
> 
> 但是，在调用 disconnect 和再次连接之后，它会多次发送相同的数据。
> 
> [98, 1, 164, 148, 56, 161, 93, 50, 0, 0, 0, 163, 64, 149, 254, 92, 141, 0, 0, 0] [98, 1, 164, 148, 56, 161, 93, 50, 0, 0, 0, 163, 64, 149, 254, 92, 141, 0, 0, 0]
> 
> [98,1,164,148,56,161,93,50,0,0,0,163,64,149,254,92,141,0,0][98,1,164,148,56,161,93,50,0,0,0,163,64,149,254,92,141,0,0]
> 
> [99, 1, 46, 54, 117, 73, 1, 35, 53, 181, 244, 9, 138, 92, 0, 0, 0, 29, 47, 15] [99, 1, 46, 54, 117, 73, 1, 35, 53, 181, 244, 9, 138, 92, 0, 0, 0, 29, 47, 15]
> 
> [99,1,46,54,117,73,1,35,53,181,244,9,138,92,0,0,0,0,29,47,15][99,1,46,54,117,73,1,35,53,181,244,9,138,92,0,0,0,29,47,15]
> 
> `// _notificationService?.value.listen((event) { if (event.isNotEmpty) { print("EVENT $event"); // so, here I got the events with the same data after reconnecting. } }); //` At the same time, if you connect to some other device, the data will be in order until we disconnect and connect again. Of course, if I close and reopen the application, I can reconnect to these devices again.
> 
> //_ 通知服务？.Listen ((EVENT){ if (EVENT.isNotEmpty){ print (“ EVENT $EVENT”) ;//so，这里我在重新连接后得到了具有相同数据的事件。});//同时，如果您连接到其他设备，数据将是有序的，直到我们断开并再次连接。当然，如果我关闭并重新打开应用程序，我可以重新连接到这些设备。
> 
> It was tested on Android and ios works same. Maybe there are some buffers not cleared in the plugin for a specific device?
> 
> 它在安卓系统上测试过，ios 系统也一样。也许某个特定设备的插件中有一些缓冲区没有被清除？

Maybe you should see this.[#317]( https://github.com/pauldemarco/flutter_blue/issues/317)

boskokg, 2022-01-31:
Please go to flutter_blue_plus and report issues there if still exist.

Hitbee-dev, 2022-03-03:
> Please go to flutter_blue_plus and report issues there if still exist.

The same problem is occurring for me.

boskokg, 2022-03-03:
@Hitbee-dev please report issue there. development on this plugin is continued there.
It would be great to investigate the problem and give PR on flutter_blue_plus

Hitbee-dev, 2022-03-04:
> @Hitbee-dev please report issue there. development on this plugin is continued there. It would be great to investigate the problem and give PR on flutter_blue_plus

I solved the problem.
We will revise the sample to float_blue_plus soon and Pull request the entire sample, so please check it.

Hitbee-dev, 2022-03-04:
I'm provides a solution. @GorIvanov
If this method is okay, please close Issues.

When we disconnected and reconnected Bluetooth, we confirmed that duplicate data came in as many times as we reconnected.
Those who contributed to solving the problem: @Hitbee-dev @K9714 @jaehyeokjeong 

We tried the following methods to solve this problem.
## 1. Identify where duplicate data comes from.
[image]

``` dart
ss.value.listen((value) async {
   //...duplicate data Issue
});
```
## 2. Check what information is received in the ```listen```
[image]

> It can be seen that the data type is StreamSubscription.
## 3. ```StreamSubscription<List<int>>? subscription = null;```
[image]

> It can be seen that the format of the data received is ```List<int>```

Therefore, it was found that subscription information could be obtained using ```StreamSubscription<List<int>>? subscription = null;``` as a global variable.
## 4. ```listen().cancel();```
[image]

> When ```listen``` was used asynchronously, it found out that if ```cancel()``` is not used, subscription information continues to be held (why duplicate data occurs).

Therefore, when the connection was terminated, subscription information was deleted using ```cancel()```
## 5. FULL CODE
[image]

I hope it helps developers who are experiencing the same problem...

The same problem occurred in flutter_blue_plus, which can be solved in this way.
If you don't mind, can I send you a PR by modifying the example code of the flutter_blue_plus you developed? @boskokg

closed 2023-06-09 by GorIvanov