pauldemarco/flutter_blue #1010: Error when add flutter_blue dependency on Windows

jianbingyan, 2022-01-07:
When I add "flutter_blue" dependency in my project with Windows OS, I have the following problems:
Cannot create link, path = 'D:\flutter\flutter\.pub-cache\_temp\dira260aa5b\macos\Classes\FlutterBluePlugin.h' (OS Error: 客户端没有所需的特权。, errno = 1314)

What should I to fix this problem?

cw2018, 2022-01-15:
解决没有，我也遇到同样的问题？

duhualong, 2022-01-17:
window 系统“设置”里面打开“开发人员模式”

ronaldmaymone, 2022-01-21:
This is a pub problem on Windows. There's an [open issue](https://github.com/dart-lang/pub/issues/1798) dealing with this. There are several fixes, none worked for me, I just moved to Ubuntu.

adarshnagrikar14, 2022-01-28:
Just Run Your IDE as an Administrator and then run "flutter pub get" in terminal. It worked for me!!