pauldemarco/flutter_blue #1011: MacOS does not prompt for Bluetooth permission

fotiDim, 2022-01-08:
MacOS does not prompt for Bluetooth permission when running through `flutter run`. I am testing it on the example app. The result is that it cannot search for devices. 

However, if I instead run the app through Xcode then it does prompt for Bluetooth permission. Any idea why running through Xcode has different behaviour?

fotiDim, 2022-01-08:
I have dug down a bit more:
❌ It seems that when running through terminal the Bluetooth permission is asked by the terminal and not from the app itself. That essentially casts the app useless as it still cannot search for devices. The same behaviour happens with VSCode where VSCode itself asks for the Bluetooth permission.

✅ However if I run the generated `.app` from finder or run through Xcode then Bluetooth permission is asked by the app as it should.

💡 I have noticed a similar behaviour with [quick_blue](https://pub.dev/packages/quick_blue) which makes me think this might be a Flutter bug.

fotiDim, 2022-01-09:
I confirmed it is a Flutter issue and opened a bug on the Flutter repo: https://github.com/flutter/flutter/issues/96369

peekpt, 2022-01-10:
@fotiDim just one off topic question: can I use a BT Dongle os a macOS app? Thanks in advance

fotiDim, 2022-01-10:
@peekpt This package uses CoreBluetooth under the hood. If your dongle has its own "management" software then I assume you need to be talking to it as a usb device. Perhaps it even has its own SDK. In that case I don't think this package can help.