pauldemarco/flutter_blue #1017: can not detect 5.2 bluetooth devices

KevinSailly, 2022-01-30:
Hi Paul,

First thank you for your great work on this plugin.

I am using your plugin for a personal project that require to detect bluetooth devices to interact with them, it was working fine until I have upgraded the bluetooth hardware to 5.2 on the external device to be detected.

I have tried many options configuring the bluetooth class and other parameters provided by hciconfig without any results, and I finally tried the flutter_bluetooth_serial from @edufolly that works well and can detect 5.2 bluetooth devices. Nevertheless, it only works on Android, which is not suitable for me.

@pauldemarco why don't you associate your efforts with @edufolly as suggested in its plugin [README](https://github.com/edufolly/flutter_bluetooth_serial) : 
_Maybe integration with flutter_blue one day ;)_

I am sure that taking the best part of your projects will lead in a great plugin for flutter community.

Regards,
Kévin

boskokg, 2022-01-31:
@KevinSailly please go to flutter_plue_plus and report issues there if still exist.

KevinSailly, 2022-01-31:
Hi @boskokg ,
Ok, great I will check it out tonight. 
Thanks,
Kévin

KevinSailly, 2022-01-31:
@boskokg ,

I have an error on while building  : 

`[  +98 ms] > Task :flutter_blue_plus:extractIncludeDebugProto
[        ] > Task :flutter_blue_plus:extractProto
[ +100 ms] > Task :flutter_blue_plus:generateDebugProto
[        ] > Task :flutter_blue_plus:javaPreCompileDebug
[        ] > Task :flutter_bluetooth_serial:generateDebugBuildConfig
[ +101 ms] > Task :flutter_blue_plus:compileDebugJavaWithJavac FAILED
[        ] An exception has occurred in the compiler (1.8.0_242-release). Please file a bug against the Java compiler via the Java bug reporting page (http://bugreport.java.com) after checking the Bug Database (http://bugs.java.com) for duplicates. Include your program and the following diagnostic in your report. Thank you.
[        ] java.lang.AssertionError: annotationType(): unrecognized Attribute name MODULE (class com.sun.tools.javac.util.UnsharedNameTable$NameImpl)
[        ] 	at com.sun.tools.javac.util.Assert.error(Assert.java:133)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations.annotationType(TypeAnnotations.java:231)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.separateAnnotationsKinds(TypeAnnotations.java:294)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.visitMethodDef(TypeAnnotations.java:1066)
[        ] 	at com.sun.tools.javac.tree.JCTree$JCMethodDecl.accept(JCTree.java:778)
[        ] 	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:49)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.scan(TypeAnnotations.java:275)
[        ] 	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:57)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.visitClassDef(TypeAnnotations.java:1042)
[        ] 	at com.sun.tools.javac.tree.JCTree$JCClassDecl.accept(JCTree.java:693)
[        ] 	at com.sun.tools.javac.tree.TreeScanner.scan(TreeScanner.java:49)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$TypeAnnotationPositions.scan(TypeAnnotations.java:275)
[        ] 	at com.sun.tools.javac.code.TypeAnnotations$1.run(TypeAnnotations.java:127)
[        ] 	at com.sun.tools.javac.comp.Annotate.flush(Annotate.java:152)
[        ] 	at com.sun.tools.javac.comp.Annotate.enterDone(Annotate.java:129)
[        ] 	at com.sun.tools.javac.comp.Enter.complete(Enter.java:512)
[        ] 	at com.sun.tools.javac.comp.Enter.main(Enter.java:471)
[   +1 ms] 	at com.sun.tools.javac.main.JavaCompiler.enterTrees(JavaCompiler.java:982)
[        ] 	at com.sun.tools.javac.main.JavaCompiler.compile(JavaCompiler.java:857)
[        ] 	at com.sun.tools.javac.main.Main.compile(Main.java:523)
[        ] 	at com.sun.tools.javac.api.JavacTaskImpl.doCall(JavacTaskImpl.java:129)
[        ] 	at com.sun.tools.javac.api.JavacTaskImpl.call(JavacTaskImpl.java:138)
[        ] 	at org.gradle.internal.compiler.java.IncrementalCompileTask.call(IncrementalCompileTask.java:74)
[        ] 	at org.gradle.api.internal.tasks.compile.AnnotationProcessingCompileTask.call(AnnotationProcessingCompileTask.java:94)
[        ] 	at org.gradle.api.internal.tasks.compile.ResourceCleaningCompilationTask.call(ResourceCleaningCompilationTask.java:57)
[        ] 	at org.gradle.api.internal.tasks.compile.JdkJavaCompiler.execute(JdkJavaCompiler.java:55)
[        ] 	at org.gradle.api.internal.tasks.compile.JdkJavaCompiler.execute(JdkJavaCompiler.java:40)
[        ] 	at org.gradle.api.internal.tasks.compile.NormalizingJavaCompiler.delegateAndHandleErrors(NormalizingJavaCompiler.java:97)
[        ] 	at org.gradle.api.internal.tasks.compile.NormalizingJavaCompiler.execute(NormalizingJavaCompiler.java:51)
[        ] 	at org.gradle.api.internal.tasks.compile.NormalizingJavaCompiler.execute(NormalizingJavaCompiler.java:37)
[        ] 	at org.gradle.api.internal.tasks.compile.AnnotationProcessorDiscoveringCompiler.execute(AnnotationProcessorDiscoveringCompiler.java:51)
[        ] 	at org.gradle.api.internal.tasks.compile.AnnotationProcessorDiscoveringCompiler.execute(AnnotationProcessorDiscoveringCompiler.java:37)
[        ] 	at org.gradle.api.internal.tasks.compile.ModuleApplicationNameWritingCompiler.execute(ModuleApplicationNameWritingCompiler.java:46)
[        ] 	at org.gradle.api.internal.tasks.compile.ModuleApplicationNameWritingCompiler.execute(ModuleApplicationNameWritingCompiler.java:36)
[        ] 	at org.gradle.api.internal.tasks.compile.CleaningJavaCompiler.execute(CleaningJavaCompiler.java:53)
[        ] 	at org.gradle.api.internal.tasks.compile.incremental.IncrementalCompilerFactory.lambda$createRebuildAllCompiler$0(IncrementalCompilerFactory.java:98)
[        ] 	at org.gradle.api.internal.tasks.compile.incremental.IncrementalResultStoringCompiler.execute(IncrementalResultStoringCompiler.java:61)
[        ] 	at org.gradle.api.internal.tasks.compile.incremental.IncrementalResultStoringCompiler.execute(IncrementalResultStoringCompiler.java:45)
[        ] 	at org.gradle.api.internal.tasks.compile.CompileJavaBuildOperationReportingCompiler$2.call(CompileJavaBuildOperationReportingCompiler.java:59)
[        ] 	at org.gradle.api.internal.tasks.compile.CompileJavaBuildOperationReportingCompiler$2.call(CompileJavaBuildOperationReportingCompiler.java:51)
[        ] 	at org.gradle.internal.operations.DefaultBuildOperationRunner$CallableBuildOperationWorker.execute(DefaultBuildOperationRunner.java:200)
[... cut ...]
           <asynchronous suspension>
           #8      run.<anonymous closure>.<anonymous closure> (package:flutter_tools/runner.dart:62:9)
           <asynchronous suspension>
           #9      AppContext.run.<anonymous closure> (package:flutter_tools/src/base/context.dart:150:19)
           <asynchronous suspension>
           #10     main (package:flutter_tools/executable.dart:94:3)
           <asynchronous suspension>
`
I am using the embedded jre with AndroidStudiounder Linux.

Regards,
Kévin

KevinSailly, 2022-01-31:
resolved after an upgrade of android studio that uses JAVA11 rather than 8.

KevinSailly, 2022-01-31:
now looking for runtime error : E/AndroidRuntime( 5095): java.lang.SecurityException: Need android.permission.BLUETOOTH_CONNECT permission for AttributionSource { uid = 10396, packageName = productif.fr.shareif_mobile, attributionTag = null, token = android.os.BinderProxy@a2d50cd, next = null }: AdapterService getRemoteName

Putting this declaration on my android manifest did not help.

KevinSailly, 2022-01-31:
By the way, I am just scanning for devices, not connecting to them, so I do not really understand this result.

boskokg, 2022-01-31:
@KevinSailly I think this is related to already reported issue with missing permission BLUETOOTH_CONNECT missing 

[](https://github.com/boskokg/flutter_blue/pull/29)

abdelrahman-abied, 2022-04-17:
try flutter_blue_plus package it's link https://pub.dev/packages/flutter_blue_plus