pauldemarco/flutter_blue #1019: PlatformException error on scan

SteMMo33, 2022-02-02:
Hi all,
I'm testing the flutter_blue plugin and I'm using version 0.8.0.
Why the GitHub main pain report 0.7.2 as lastest release?

Anyway when I start with the scan procedure I have the following error:

`
E/flutter (13489): [ERROR:flutter/lib/ui/ui_dart_state.cc(209)] Unhandled Exception: PlatformException(error, Field devices_ for b.b.a.u not found. Known fields are [private b.a.a.b0$i b.b.a.u.e, private static final b.b.a.u b.b.a.u.f, private static volatile b.a.a.a1 b.b.a.u.g], null, java.lang.RuntimeException: Field devices_ for b.b.a.u not found. Known fields are [private b.a.a.b0$i b.b.a.u.e, private static final b.b.a.u b.b.a.u.f, private static volatile b.a.a.a1 b.b.a.u.g]
E/flutter (13489): 	at b.a.a.v0.a()
E/flutter (13489): 	at b.a.a.v0.a()
E/flutter (13489): 	at b.a.a.v0.a()
E/flutter (13489): 	at b.a.a.k0.a()
E/flutter (13489): 	at b.a.a.k0.a()
E/flutter (13489): 	at b.a.a.d1.a()
E/flutter (13489): 	at b.a.a.d1.a()
E/flutter (13489): 	at b.a.a.z.m()
...`

Any idea?
Thanks

d-wolf, 2022-02-04:
Had the same problem. This worked for me: https://github.com/pauldemarco/flutter_blue/issues/662#issuecomment-741710564.

SteMMo33, 2022-02-08:
Thanks, I can confirm that solved my problem.

closed 2022-02-08 by SteMMo33