pauldemarco/flutter_blue #1030: macos build only finds devices when running the .app

robnorr, 2022-02-20:
Hi all,

Using the example app from this repo, when I run `flutter run` from the terminal or from VSCode, the app runs and it gets access to the bluetooth hardware, but the "Find devices" list is empty. If I run the build app bundle (in build/macos/Build/Products/Debug/) it runs, and the Find devices list gets populated. However, since I'm running the app this way, I no longer have access to hot reload etc.

I've modified the info.plist, and I've done the step described here https://github.com/pauldemarco/flutter_blue/issues/788, but I just can't figure out what's going wrong.

Is there a way to either 
a) run the app bundle directly but still have access to hot reload etc.
b) run `flutter run` and have the bluetooth devices get found?

Thanks for your help.

robnorr, 2022-02-20:
I think the issue is likely related to this https://github.com/flutter/flutter/issues/96369. I hope it gets fixed soon

dirablue, 2022-07-26:
@robnorr 
> If I run the build app bundle (in build/macos/Build/Products/Debug/) it runs, and the Find devices list gets populated

this is good information for testing BLE for now.
thanks.