pauldemarco/flutter_blue #1034: detecting end of device scan

letisoft, 2022-03-07:
Hello,

code as simple as it is:
`    var subscription = FlutterBlue.instance.scanResults.listen((results) {
      // do something with scan results
      for (ScanResult r in results) {
        print('${r.device.name} found! rssi: ${r.rssi}');
      }
    })
    ..onError((err){
      print(err.toString());
    })
    ..onDone(() {
      print("done");
    })
    ;
`

all works but on done is never called, so I am not able to detect when scanning has finished?

Thanks!
V

watsmyname, 2022-03-09:
This may help
```
scanOnProgress = false;
scanForDevices(){ //method that executes
//  FlutterBlue.instance.scanResults .....
}
isScanning();
```

```
isScanning(){
  FlutterBlue.isScanning.listen((event){
     setState(() {
         scanOnProgress  = event;
     });
  });
}
```
Then you can check value of `scanOnProgress` , to determine whether scanning is completed or not.

letisoft, 2022-03-10:
Thanks, not perfect but does the job