pauldemarco/flutter_blue #1038: heart rate measurement characteristics not read

rlfgh2002, 2022-03-15:
I used heart rate data for mi band4, 5
"00002a37-0000-1000-8000-00805f9b34fb" this uuid can be read on iOS, but not on Android.

Android was this characteristics.properties.read = false 
but iOS return true
what should i do to solve this problem?

this is my code :

    for (blep.BluetoothCharacteristic c in _service!.characteristics) {
      String uuid = "0x${c.uuid.toString().toLowerCase().substring(4, 8)}";
      if (c.properties.read) {
        if (c.isNotifying) {
          await c.setNotifyValue(true);
        }
        c.read();
        if (uuid == "0x2a37") {
          debugPrint(
              ":::::::::: Heart Rate Measurement => ${c.lastValue[1]} ::::::::::");
        } else if (uuid == "0x2a39") {
          debugPrint(
              ":::::::::: Heart Rate Control Point => ${c.lastValue[1]} ::::::::::");
        }
      }
    }

closed 2022-03-15 by giroskii

brutalzinn, 2022-11-24:
how did you solve this problem?

giroskii, 2022-11-24:
@brutalzinn 
i solved this problem used try ~ catch
If a catch error appears, it is already connected, and if there is no error, it is judged as disconnect.

giroskii, 2022-11-24:
> how did you solve this problem?

@brutalzinn
i solved this problem used try ~ catch
If a catch error appears, it is already connected, and if there is no error, it is judged as disconnect.