pauldemarco/flutter_blue #1040: Scan not working

jatin-web, 2022-03-22:
Scan is not working now

FXschwartz, 2022-03-24:
@jatin-web You will need to provide a lot more information about your use case.

What kind of device are you using, what does your scan code logic look like, what error messages are you getting.

Currently, scanning is working for me.

JakubRybakowski, 2022-03-30:
you need permissions for localisation

    await Permission.locationWhenInUse.request();
    await Permission.locationAlways.request();

    Location location = new Location();

    bool _serviceEnabled = await location.serviceEnabled();
    if (!_serviceEnabled) {
      _serviceEnabled = await location.requestService();
      if(!_serviceEnabled) {

      }
    }

isysenko, 2022-03-31:
@JakubRybakowski thanks)