pauldemarco/flutter_blue #1044: Long read/write support

ysba, 2022-03-31:
Hello. I need to transfer data via BLE with length larger than the maximum allowed MTU size (512). I know that BLE supports long read or write operations, whose has a write prepare command so the other device can prepare to receive a larger buffer splitted in chuks of MTU size. Are these long read and write operations supported by flutter_blue? In the write method listed in flutter_blue API documentation doesn't seem to have such option. Thank you.

abdelrahman-abied, 2022-04-17:
try flutter_blue_plus package it's link https://pub.dev/packages/flutter_blue_plus

explorer2050, 2023-06-14:
@abdelrahman-abied  does flutter_blue_plus package support Long read/write ?