pauldemarco/flutter_blue #1053: I can't disconnect the device from the connected device if the app restarts

syahrulafrizal, 2022-04-11:
■ Scenario 0
1. connect to some device
2. disconnect above device
-----> doing well

■ Scenario 1
1. connect to some device
2. terminate app
3. restart app
-----> we can find connected device by
-----> listening connectedDeviceStream

How can i disconnect device what found by connectedDeviceStream
not from manually connectToDevice nor connectToAdvertisingDevice ?

i already know canceling stream from connectToDevice
or connectToAdvertisingDevice take disconnection

but, connected device from connectedDeviceStream have no stream
so i can't disconenct those devices..... T_T

■ Trying
after Scenario 1, if i found already connected device
then make connection and cancel...

but can't get disconnected callback by ConnectionStateUpdate
(both stream from connection and FlutterReactiveBle.instants's connectedDeviceStream)

also can't scan that device...

abdelrahman-abied, 2022-04-17:
try flutter_blue_plus package it's link https://pub.dev/packages/flutter_blue_plus

maddionatoplus, 2022-07-25:
@syahrulafrizal hi,
did you find a solution using FlutterBlue?

syahrulafrizal, 2022-07-25:
I try using flutter_blue_plus, but the issue is still the same
and i use bluetooth turn off function to disconnect bluetooth connection
and the issue solved, bluetooth is discomnected

[image: image.png]

syahrulafrizal, 2022-07-25:
onDisconnectBluetooth(BluetoothDeviceModel data) async {
printInfo(info: "onDisconnectBluetooth");
if (Get.isDialogOpen!) {
Get.back();
}
listDeviceConnected.clear();
isConnectedBluetoothDevice = false;
isConnectedNetworkDevice = false;
isLiveStreamDevice = false;
bluetoothDeviceName = "...";
networkDeviceName = "...";
streamUrl = "";
batteryLevel = "0%";
batteryCharging = 0;
deviceConnected = null;
update();
FlutterBluePlus.instance.turnOff();
}

syahrulafrizal, 2022-07-25:
> @syahrulafrizal hi, did you find a solution using FlutterBlue?

I try using flutter_blue_plus, but the issue is still the same
and i use bluetooth turn off function to disconnect bluetooth connection
and the issue solved, bluetooth is discomnected

onDisconnectBluetooth(BluetoothDeviceModel data) async {
    printInfo(info: "onDisconnectBluetooth");
    if (Get.isDialogOpen!) {
      Get.back();
    }
    listDeviceConnected.clear();
    isConnectedBluetoothDevice = false;
    isConnectedNetworkDevice = false;
    isLiveStreamDevice = false;
    bluetoothDeviceName = "...";
    networkDeviceName = "...";
    streamUrl = "";
    batteryLevel = "0%";
    batteryCharging = 0;
    deviceConnected = null;
    update();
    FlutterBluePlus.instance.turnOff();
  }

maddionatoplus, 2022-07-25:
Thank you, I don't like turning off the user's Bluetooth very much.

figureai, 2022-08-03:
Have any solvetion？

maddionatoplus, 2022-08-03:
As a solution, I consider the connected Devices as available devices. If the user needs to connect to a connected device I call connect method on it, otherwise, if the user needs to connect to another device I look for other scanned devices.

figureai, 2022-08-03:
but，When the phone is connected to Bluetooth, calling device.connect() method will report an error

abdelrahman-abied, 2022-08-05:
flutterBlue.connectedDevices.then((connectedDevices) {
      if (connectedDevices.isNotEmpty) {
        connectedDevices.map((device) async {
          return await device.disconnect();
        });
      }
    });
// connect to required device
 requiredDevice.connect();

a1573595, 2022-08-25:
> ■ 場景0
> 
> 1. 連接到某個設備
> 2. 斷開上面的設備
>    -----> 做得很好
> 
> ■ 場景一
> 
> 1. 連接到某個設備
> 2. 終止應用程序
> 3. 重新啟動應用程序
>    -----> 我們可以通過
>    -----> 監聽 connectedDeviceStream找到連接的設備
> 
> 如何斷開已連接設備流找到的設備， 而不是手動 connectToDevice 或 connectToAdvertisingDevice ？
> 
> 我已經知道從 connectToDevice 取消流 或 connectToAdvertisingDevice 斷開連接
> 
> 但是，來自 connectedDeviceStream 的連接設備沒有流 ，所以我無法斷開這些設備.....T_T
> 
> ■ 在場景 1 之後嘗試，如果我發現已經連接的設備， 則建立連接並取消...
> 
> 但無法通過 ConnectionStateUpdate 獲得斷開的回調 （來自連接的流和 FlutterReactiveBle.instants 的 connectedDeviceStream）
> 
> 也無法掃描該設備...

Same issue on iOS, but Android will disconnect when the app is closed.

yudeying918, 2022-12-23:
> flutterBlue.connectedDevices.then((connectedDevices) { if (connectedDevices.isNotEmpty) { connectedDevices.map((device) async { return await device.disconnect(); }); } }); // connect to required device requiredDevice.connect();

Do u mean to add this code in dispose() method?

jimin5329060, 2023-04-24:
i call flutterBlue.connectedDevices.then  android is return device list but ios still return empty