pauldemarco/flutter_blue #1054: Upgrade Dependencies

yusmi25, 2022-04-18:
hello, thanks to the Developer to make this package, It's very useful when working with Bluetooth so far, but any plan to upgrade this package, because the Dependencies are too old.
Thanks 👍 💯

kunalandroid, 2022-04-18:
> hello, thanks to the Developer to make this package, It's very useful when working with Bluetooth so far, but any plan to upgrade this package, because the Dependencies are too old. Thanks 👍 💯

I totally agree. @pauldemarco sir would this package be upgraded as I face some of the API deprecation warnings while building. It's an awesome package and I don't wanna loose it. Thanks <3

AguSampietro, 2022-04-22:
This package hasn't been maintaining lately. I suggest you to migrate to flutter_blue_plus.