pauldemarco/flutter_blue #1055: App crash while enabling BLE on Google Pixel 3a XL running Android 12

parag-kulkarni, 2022-04-20:
Flutter Blue works fine upto Android 11. The crash is specific to android 12.
Flutter version used is 2.8. PFA the crash log.

[android12FlutterBlueErrorSheet.txt](https://github.com/pauldemarco/flutter_blue/files/8518433/android12FlutterBlueErrorSheet.txt)
1

AguSampietro, 2022-04-22:
I had kind of the same issue, but this package hasn't been maintaining lately. I suggest you to migrate to flutter_blue_plus.

anik786, 2022-06-17:
Had the same issue. https://github.com/boskokg/flutter_blue_plus solves the issue.

Also use new Android permissions:

```
    <!-- @link https://developer.android.com/guide/topics/connectivity/bluetooth/permissions -->
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission
        android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission
        android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />
    <uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <!-- Request newer Bluetooth permissions for newer devices -->
    <uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
```