pauldemarco/flutter_blue #1059: No devices are showing on my phone, Samsung S10e, Android 12

MoaadLala, 2022-05-01:
I have a problem with the package, i guess that the problem is with my phone or the Android 12 is not yet supported, because it works fine with other devices.
I was just wondering whether am doing something wrong, or if i can do something to make it work on my phone...
Thanks in advance

thusithadeepal9356, 2022-05-02:
could you provide the github repo of your project?
Is you try to connect with another phone device?

MoaadLala, 2022-05-02:
am using the example app, and yes am trying to connect to another mobile device, i want to send files to it

boskokg, 2022-05-02:
@MoaadLala please close the issue and use flutter_blue_plus

thusithadeepal9356, 2022-05-03:
Why is this not supported?

anik786, 2022-06-17:
Had the same issue. https://github.com/boskokg/flutter_blue_plus solves the issue.