pauldemarco/flutter_blue #1069: Error in write and notify characteristics platform exception

anas-iqbal, 2022-05-25:
Hello,

I have been facing the same issue with this library flutter blue.

```
await setNotifyValue(true); // exception: set_notification_error, error when writing the descriptor, null, null

await writeChar.write(cmd); // exception: PlatformException(write_characteristic_error, writeCharacteristic failed, null, null).
```

I am not running this lines concurrently I have used await. I am working with 3 BLE devices,  it works most of them time but sometimes it start getting this exception and my app misbehaves and this error can happen in any of the device not even specific to any device. any idea?

johananh, 2022-05-25:
this works for me ( try up to 3 times to write)
----------------------------------------------------------
 Future<int> writeChar(Guid uuid, List<int> value) async {
    var c = getChar(uuid);
    if (c != null) {
      for (int i = 0; i < 3; i++) {
        try {
          await c.write(value);
          return i + 1;
        } catch (e) {
          sleep(const Duration(milliseconds: 100));
        }
      }
      throw Exception('Error Writing Characteristic');
    }
    return 0;
  }

alexvnesta, 2023-02-26:
@johananh this worked for me. Thank you!!!