pauldemarco/flutter_blue #1073: How to read more than 20 bytes data (MTU negotation is not available)

aaronchen96, 2022-06-01:
Hello,

I have a problem to read the proper amount of data sent by BLE peripheral. The peripheral cannot send bigger packages than 20 bytes (MTU is not negotiable). It cuts every message into 20 bytes packets. Unfortunately, I am not able to catch the first few packets, just the last one. While i subscribe to a characteristic and force it to send data, I can't find a way to catch the single packets. I met similar problem on pure Android app, but there I was able to save each packet from a characteristic and then reuse it. Here, it seems like they disappear and only the last one is used. Any ideas in this case?
I will be very grateful for any insights.

closed 2022-06-01 by aaronchen96

kennetgo, 2022-06-14:
Hello, how did you solve that issue? I hope you can help me. Thanks :)

aaronchen96, 2022-06-14:
@kennetgo 
You have to subscribe to the characteristic instead of performing just a casual read. When the peripheral will start transferring data, your subscribed characteristic will receive each packet, instead of just the last one which is saved in it's memory.

"While i subscribe to a characteristic and force it to send data, I can't find a way to catch the single packets." - You can make yourself a global variable, to which every data transferred would be added. Remember to clean the global variable every time you want to receive new data. That's just one way of "bypass".

Also, i tried to change the MTU size of the device
`   await connectedDevice!.requestMtu(250);
              var mtuChanged = Completer<void>();
              // mtu is of type 'int'
              var mtuStreamSubscription = connectedDevice!.mtu.listen((mtu) {
                if (mtu == 250) mtuChanged.complete();
              });
              // await mtuChanged.future; // set timeout and catch exception
              mtuStreamSubscription.cancel();`

kennetgo, 2022-06-14:
Thanks Aaron! I did exactly what you said and for some reason I get all the packages and sometimes just the last one, it works randomly. Do you know what can be happening?