pauldemarco/flutter_blue #1076: Read Signals from BLE devices

Alade5673, 2022-06-11:
How can I read signals from my connected BLE device using flutter blue?

blacktorch, 2022-06-13:
Do you mean iBeacon signals?

Alade5673, 2022-06-13:
no

a ble device was built that once connected to, sends a 43 digits response.

How can I read the response immediately after connecting to the device using flutter blue

Alade5673, 2022-06-13:
@blacktorch 

no

a ble device was built that once connected to, sends a 43 digits response.

How can I read the response immediately after connecting to the device using flutter blue

blacktorch, 2022-06-13:
Are you creating any gatt service? and if so do you have a write or notification characteristics defined?

Alade5673, 2022-06-13:
@blacktorch 

It's a smart home device.

blacktorch, 2022-06-13:
Check its documentation to see if it is running any GATT service.

blacktorch, 2022-06-13:
List<BluetoothService> services = await device.discoverServices();
services.forEach((service) {
    // do something with service
});

Alade5673, 2022-06-14:
@blacktorch 

[image]

I use a BLE Terminal app to connect to the device, and immediately after connection, I get a response values.

So my challenge is that, after I connected via flutter blue to the device, how do I retrieve the response value?