pauldemarco/flutter_blue #1077: In the Android 12 environment, the plug-in still reports an error that the Bluetooth permission cannot be obtained when the permission is obtained

1271215069, 2022-06-13:
In the Android 12 environment, the plug-in still reports an error that the Bluetooth permission cannot be obtained when the permission is obtained
[image]
[image]

anik786, 2022-06-17:
Had the same issue. https://github.com/boskokg/flutter_blue_plus solves the issue.

kdjun97, 2022-06-20:
I think u just use above plugin flutter_blue_plus
I tested the flutter_blue_plus ^1.1.3 version in Android 11, 12 Version for BLE Connection.
And, I checked that it worked well.

Apply the plugin and just change the code.
ex)
FlutterBlue flutterBlue = FlutterBlue.instance ->FlutterBluePlus flutterBlue = FlutterBluePlus.instance;

That's all.

vricosti, 2022-07-12:
I have used flutter_blue_plus but finally I put back flutter_blue because it works fine on android 12 as long as you ask for permissions:

```
Future<int> getAndroidSdk() async {
    DeviceInfoPlugin deviceInfo = DeviceInfoPlugin();
    AndroidDeviceInfo androidInfo = await deviceInfo.androidInfo;
    return androidInfo.version.sdkInt ?? 30;
  }

  Future<bool> requestBluetoothAccess() async {
    Future<bool> permStatus;
    if (Platform.isAndroid && await getAndroidSdk() > 30) {
      permStatus = _requestAccess(Permission.bluetoothScan)
          .then((isGranted) => _requestAccess(Permission.bluetoothConnect))
          .then((isGranted) => isGranted);
    } else {
      permStatus = _requestAccess(Permission.bluetooth);
    }
    return permStatus;
  }
```
and I am using the following manifest:

```
<!-- New Bluetooth permissions in Android 12 -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"/>
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:maxSdkVersion="30" android:name="android.permission.BLUETOOTH"/>
    <uses-permission android:maxSdkVersion="30" android:name="android.permission.BLUETOOTH_ADMIN"/>

    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_WIFI_STATE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.MANAGE_EXTERNAL_STORAGE" tools:ignore="ScopedStorage" />
    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```