pauldemarco/flutter_blue #1080: RN41XVv bluetooth module

ebelouet, 2022-06-21:
HI,

Is it compatible with this bluetooth module of microchip ?

Sincerely
eric

sjrdvndwg, 2022-06-21:
as far as I know that chip doesnt use bluetooth low energy and wont be compatible. 
how ever [this ](https://github.com/edufolly/flutter_bluetooth_serial) might work.

sincerely
Sjoerd

ajsb85, 2022-07-02:
Hi, @ebelouet 

Android should support it. But not iPhone. iOS support only BLE devices.

Some references of that module:

# RN41XV
### Status: Not Recommended for new designs

The RN41XV module provides drop-in, certified Bluetooth connectivity for existing systems using 802.15.4 modules.  Based on the popular 2 x 10 (2mm) socket footprint often found in embedded applications, the RN41XV offers a complete wireless solution for customers looking to migrate to a standard protocol without modifying existing hardware.  The RN41XV is built upon Microchip's RN41 low power Bluetooth module.  The module has an embedded Bluetooth stack and supports multiple interface protocols and profiles including the commonly used SPP and HID profiles.

- Fully certified Bluetooth® version 2.1 module, supports version 2.1 + Enhanced Data Rate (EDR)
- Backwards-compatible with Bluetooth version 2.0, 1.2, and 1.1
- Pin compatible with widely used 2 x 10 (2mm) socket typically used for 802.15.4 applications
- Low power: 30 mA connected, < 10 mA sniff mode
- UART (SPP or HCI) and USB (HCI only) data connection interfaces
- Sustained SPP data rates: 240 Kbps (slave), 300 Kbps (master)
- HCI data rates: 1.5 Mbps sustained, 3.0 Mbps burst in HCI mode
- Embedded Bluetooth stack profiles included (requires no host stack): GAP, SDP, RFCOMM, and L2CAP protocols, with SPP, HID and DUN profile support
- Bluetooth SIG certified
- Certifications: FCC, IC, CE
- Environmentally friendly, RoHS compliant
- 2 antenna options available: Chip antenna (RN41XVC-I/RM)and U.FL connector for external antenna (RN41XVU-I/RM)