pauldemarco/flutter_blue #1105: How can i store Bluetooth device and connect after ?

InfernoSnake, 2022-11-01:
Hi,

I am developing a Bluetooth app which can be connect different devices. My problem is i cant store Bluetooth device data on mobile storage(i tried to use hive,sharedpreferences). I searched but i guess the package cant support this feature. Could you help me with another way ?

What i want --> 
1) Search devices
2) connect one of them
3) Close the app and relaunch it.
4) there is a list that earlier connected devices.
5) select one of them and connect it.

bernier, 2022-12-20:
I was looking for the same thing. Have you found a solution?
Right now, I have to go to the phone settings, forget the device, and then I'm able to pair it again, which is pretty bad.

nairboon, 2022-12-27:
You could store such a list using shared_preferences.
But consider using [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus), because flutter_blue is no longer maintained.

karthikmohan4, 2023-01-05:
I think you can have only the records of previously connected device only as name(String).
And if you relaunch the app. The bluetooth screen will be initialized and also the device should be available to connect. Why are you trying to select from prevoius & connect ?