pauldemarco/flutter_blue #1109: Unhandled Exception: MissingPluginException(No implementation found for method startScan on channel flutter_blue_plus/methods)

RafCoder90, 2022-11-08:
I entered in the pubspec.yaml the dipendence flutter_blue_plus: ^1.3.1 .
I have a button that calls a method. In this method I declare a FlutterBlue instance but:
Unhandled Exception: MissingPluginException(No implementation found for method startScan on channel flutter_blue_plus/methods)

iamsupercode, 2022-11-18:
Have you tried rebuilding your app ?

RafCoder90, 2022-11-18:
yes

baranacikgoz, 2022-11-19:
The same error sometimes occurs when the application is in background for me

RafCoder90, 2022-11-19:
E/flutter ( 8557): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method setLogLevel on channel flutter_blue_plus/methods)
E/flutter ( 8557): #0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:294:7)
E/flutter ( 8557): <asynchronous suspension>

saguado69, 2023-05-24:
Same issue after upgrading from 1.4.0 that was working fine on IOS and now, with 1.5.2 does not work anymore ...

iamsupercode, 2023-05-24:
run flutter clean and try again

Gishobert Gwenzi

Full Stack Software Developer 

p:  +27104439799

22 Magwa Crescent, Gateway West, Waterfall City  Midrand    2066

petalmafrica.com

+27115848290 +260630020001 +2638677240191

Facebook Twitter LinkedIn

saguado69, 2023-05-24:
Thanks, now I have a Missing Plugin Implementation on those two lines that worked fine before upgrading:

    // CHECK AND AWAIT FOR BLE UNTIL IS ON
    await ble.isOn.then((value) async =>
        // CHECK AND AWAIT FOR BLE UNTIL IS AVAILABLE
        await ble.isAvailable.then((value) async {