pauldemarco/flutter_blue #1118: Filter devices in flutter_blue

hhcng, 2023-01-01:
I use flutter_blue to scan ble devices for social distancing app. However I'm not sure if what I did is correct. Since I make it general on scanning devices, I cannot know their characteristic and all I know is I only need to scan phone devices. Before I added the filtering which I use 'where' while scanning, it's functioning well even though it scan not just mobile devices but all devices. After I added them, it does not working at all even the floatingbutton to tap to scan devices. 

Here is my code before I add the filtering:

FlutterBlue flutterBlue = FlutterBlue.instance;

  List<ScanResult> scanResultList = [];
  bool _isScanning = false;

  @override
  initState() {
    super.initState();
    // reset bluetooth
    initBle();
  }

  void initBle() {
    // listener to get BLE scan status
    flutterBlue.isScanning.listen((isScanning) {
      _isScanning = isScanning;
      setState(() {});
    });
  }

  /*
  scan start or stop function
  */
  scan() async {
    if (!_isScanning) {
      // if not scanning
      // Delete the previously scanned list
      scanResultList.clear();
      // Start scanning, timeout 4 seconds
      flutterBlue.startScan(timeout: Duration(seconds: 4));
      // scan result listener
      flutterBlue.scanResults.listen((results) {
        // List<ScanResult> Copy the results value of the form to scanResultList
        scanResultList = results;
        // UI renewal
        setState(() {});
      });
    } else {
      // Stop scanning if scanning
      flutterBlue.stopScan();
    }
  }

And this is my code after I tried to filter them:

FlutterBlue flutterBlue = FlutterBlue.instance;

  List<ScanResult> scanResultList = [];
  bool _isScanning = false;

  @override
  initState() {
    super.initState();
    // reset bluetooth
    initBle();
    NotificationWidget.init();
  }

  void initBle() {
    // listener to get BLE scan status
    flutterBlue.isScanning.listen((isScanning) {
      _isScanning = isScanning;
      setState(() {});
    });
  }

  /*
  scan start or stop function
  */
  scan() async {
    if (!_isScanning) {
      // if not scanning
      // Delete the previously scanned list
      scanResultList.clear();
      // Start scanning, timeout 4 seconds
      final scanResult = FlutterBlue.instance.scan().where((scanResult) {
        // Filter devices by manufacturer data and local name
        final advertisingData = scanResult.advertisementData;
        final deviceName = scanResult.device.name;
        return (advertisingData.manufacturerData[0x004C] == [0x02, 0x15] ||
            advertisingData.manufacturerData[0x0051] != null) &&
            deviceName.startsWith('Phone');
      });
      // scan result listener
      flutterBlue.scanResults.listen((results) {
        // List<ScanResult> Copy the results value of the form to scanResultList
        scanResultList = results;
        // UI renewal
        setState(() {});
      });
    } else {
      // Stop scanning if scanning
      flutterBlue.stopScan();
    }
  }

karthikmohan4, 2023-01-05:
It's simple as that. First scan the devices and add it in a List. Now create another list on filtering the existing list.
Once you have list of filtered device you will be having the device properties and use that to discover the services after connecting. it will have multiple services & characteristics. Now create a characteristic vairiable and assign the device characteristic to it. Now you're ready to make use of the characteristics and perform read/write or any operatin

EliaTolin, 2023-08-06:
Hello, this plugin is obsolete and maintenance is stopped.
Use the continuous and active package : [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus)