pauldemarco/flutter_blue #1123: Can not connect a device with Android

mehmettop, 2023-01-24:
Hello All,

I am facing a connection issue for Android devices. The issue is, while everything is working perfectly on iOS side, on Android, I can not connect to device. There is no issue with scanning and getting device advertisement data. However when try to connect the device after scanning, application give me the 133 status error. The strange thing, in connect phase I could not get any operation system prompt that inform you about pairing.

I've trying everything regarding permission (Android 12) and runtime permission... :/

Do you have any recommendation to solve the problem or is there anyone who faced the same problem?

Thank you,
Best Regards.

BrietGame, 2023-01-27:
I have same issue.

TongCong12138, 2023-02-22:
`await _device?.connect(autoConnect: false, timeout: const Duration(seconds: 10));`

将连接方法的autoConnect设置为false

dulifu, 2023-05-08:
I have same issue.did you solve it？ @mehmettop