pauldemarco/flutter_blue #1129: minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue]

andres7guillen, 2023-04-14:
I'm getting this error: minSdkVersion 16 cannot be smaller than version 19 declared in library [:flutter_blue], I know what I have to do is add:
 defaultConfig {
        minSdkVersion 19
        }
I added this line in the build.gradle file but I'm still getting the same error, also I have in my pubspec.yaml my dependencie like this:
dependencies:
  flutter:
    sdk: flutter
  cupertino_icons: ^1.0.2
  flutter_blue: ^0.8.0
  
  I don't know if I'm missing something.

peekpt, 2023-04-17:
if you want to use  flutter 3.x, use flutter_blue_plus instead

andres7guillen, 2023-04-17:
Ok I'm gonna take a look, thanks !!

EliaTolin, 2023-08-06:
Hello, this plugin is obsolete and maintenance is stopped.
Use the continuous and active package : [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus)