pauldemarco/flutter_blue #1132: There is a problem compiling the library with Flutter 3.7.12 with Gradle 8

tonisole, 2023-04-26:
When you try to compile the project the flutter_blu library fail.

_distributionUrl=https\://services.gradle.org/distributions/gradle-8.0.2-all.zip or gradle-8.1.1-all.zip
static String ndkVersion = "25.1.8937393"
classpath 'com.android.tools.build:gradle:8.0.0'_

**FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_blue'.
> Could not get unknown property 'source' for generate-proto-generateDebugProto of type org.gradle.api.internal.file.DefaultSourceDirectorySet.**

AndreaMinato, 2023-06-26:
Did you managed to get it to build?

Binozo, 2023-07-29:
Same problem here

EliaTolin, 2023-08-06:
Hello, this plugin is obsolete and maintenance is stopped.
Use the continuous and active package : [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus)

pauldemarco, 2023-08-06:
@EliaTolin  this plug-in is still used internally and externally by active products. Please avoid spamming GitHub issues.