pauldemarco/flutter_blue #1136: Race condition between Connect timeout and a successful connection leaving active connection stranded

jasaw, 2023-06-14:
If a Bluetooth LE connection is established while the connect timeout is triggering, flutter_blue will think that the connect call has timed out and failed but the Android OS still maintains the Bluetooth connection. Calling disconnect on flutter_blue does not disconnect the BLE connection.

The best way to trigger this bug is to set connect timeout to 4 seconds, and run hundreds of bluetooth connect and disconnect to a device that has weak signal strength (-90dB to -95dB). I have tested this on Android 8.

EliaTolin, 2023-08-06:
Hello, this plugin is obsolete and maintenance is stopped.
Use the continuous and active package : [flutter_blue_plus](https://github.com/boskokg/flutter_blue_plus)

chipweinberger, 2023-08-09:
This is fixed in https://pub.dev/packages/flutter_blue_plus