pauldemarco/flutter_blue #1145: ITMS-90863: Apple silicon Macs support issue

AlexBacich, 2023-09-20:
I've got this email when I am publishing app to AppStore (or TestFlight). 

How may I solve it? 

> Dear Developer,

We identified one or more issues with a recent delivery for your app, *****. Your delivery was successful, but you may wish to correct the following issues in your next delivery:

ITMS-90863: Apple silicon Macs support issue - The app uses symbols that are not present on Mac:

@rpath/flutter_blue_plus.framework/flutter_blue_plus
_OBJC_CLASS_$_FlutterError
After you’ve corrected the issues, you can upload a new binary to App Store Connect.

Best regards,

The App Store Team

EXEIdeas, 2023-09-23:
Yes, I am getting also the same error while developing and running in device...

```Installing and launching...
(lldb) dyld[1764]: Symbol not found: (_OBJC_CLASS_$_FlutterError)
  Referenced from: '/private/var/containers/Bundle/Application/40C4C70F-AB52-4CCE-A35E-4A576270B81D/Runner.app/Runner'
  Expected in: '/private/var/containers/Bundle/Application/40C4C70F-AB52-4CCE-A35E-4A576270B81D/Runner.app/Frameworks/flutter_blue.framework/flutter_blue'
* thread #1, stop reason = signal SIGABRT
    frame #0: 0x0000000105e0b2cc dyld`__abort_with_payload + 8
dyld`:
->  0x105e0b2cc <+8>:  b.lo   0x105e0b2e8               ; <+36>
    0x105e0b2d0 <+12>: stp    x29, x30, [sp, #-0x10]!
    0x105e0b2d4 <+16>: mov    x29, sp
    0x105e0b2d8 <+20>: bl     0x105dd164c               ; cerror_nocancel
Target 0: (Runner) stopped.
Error launching application on Salman’s phone .

vinicz, 2023-10-03:
Found a hint here https://developer.apple.com/forums/thread/735610, the main cause is that the iOS code still catches and throws FlutterErrors. After we got rid of that in our fork everything worked again. Here is the commit: https://github.com/wearnotch/flutter_blue/commit/4a0cad8ac8417a4460b4230a9429f58276696422 (sorry there are a lot of indentation changes as well)

chipweinberger, 2023-10-05:
update to latest flutter_blue_plus and it will resolve this issue